OmNomNomagon/ReDuplicator-MK3x #1: X Gantry Cable Box Bottom Poor Side Wall Extrusion

safetychief8, 2018-01-16:
There's poor extrusion in the walls of the X gantry cable box bottom. I loaded into S3D, and output the gcode file. The other items are printing without issue. Please check to see if there's an issue with the code. Tried to print it 3 times with tweaks in the settings, but had the same result.

OmNomNomagon, 2018-01-17:
In S3D what are your thin wall behavior settings?
I've got it setup as:

External Thin Wall: Perimeters Only
Internal Thin Wall: Allow Gap Fill
Allowed Perimeter Overlap: 20% 

Dan you give me a more detailed description or a pic? Does it show up in the layer previews? I checked, it looks okay in mine but If it persists I'll bump the wall thicknesses by 0.5mm, which should fix it up.

With regards to washers, the only part that may need them is the X motor mount, and only if you use neoprene washers to try to decouple the motor from the frame. I'll update the wiki to clarify.

safetychief8, 2018-01-17:
Hello,

With regards to printing, it only seems to be that one part for some reason, all the others print fine, that's why I suggested that that code or print settings may be suspect. I'll look to purchase the washers for the appropriate sized screws, thanks.

Sent via the Samsung Galaxy S8, an AT&T 4G LTE smartphone

safetychief8, 2018-01-19:
I've attached some pictures of the extruded part.

[image]
[image]
[image]

OmNomNomagon, 2018-01-21:
Oh wow, don't think its an issue with the file. Is this PETG? It looks like maybe you need to dial up the temps a bit and maybe increase your extrusion modifier a tad.

safetychief8, 2018-01-22:
It’s PLA, all the other parts extruded ok with the same material and settings.

OmNomNomagon, 2018-01-22:
[image]

I printed another one out super quick with no issues. 
Id say this is an issue with your printer that is only showing up on parts with thinner or finer details. I used to have a similar problem with certain types of petg. Without knowing more about your setup, reckon you should check the following.
- Double check your extruder drive gear is not loose or clogged,
- is your fan aimed to high? if the part fan is cooling your heater block it could cause these sorts of issues. Try reducing fan speed.
- double check your esteps https://3dprinterwiki.info/extruder-steps/
- retraction might be too high, causing melted plastic to move too far past the heatbreak, It would cause issues specifically on thinner prints with more retractions. try reducing this.
- Bump up your temperature 
- Bump up your extrusion multiplier
- might just be bad or inconsistent filament

Please let me know if you have any luck with this.

safetychief8, 2018-01-23:
Thanks, I'll try them, just set up the Maker V2, I'll try it on that printer instead of the delta.

closed 2018-01-29 by OmNomNomagon