OmNomNomagon/ReDuplicator-MK3x #9: Assembling the fan with the titan aero extruder

pgrandin, 2020-03-05:
I am having some issues figuring out how to properly assemble the fan shroud onto the X carriage backplate. I somewhat got it to hold but it's dangling a bit and it's often scratching my prints causing issues.

The wiki at https://github.com/OmNomNomagon/ReDuplicator-MK3x/wiki/5.2-Hotend-Titan-Aero-Upgrade says 
> Use 2 16mm M3 screws through the bottom x carriage screw holes and into this part.

It does not mention if we should use nuts on side, or from which side the screws should be inserted (I think that it is from the direction in the screenshot below as otherwise the intake is blocking the path of one of the screw).

Using 16mm screws without nuts doesn't really hold the shroud in place, but on the shroud there is no hole shaped like a nut to help secure it in place, and it is quite difficult to insert one there and hold it in place. 

What is the right way to do it? This section view should help visualize the issue i'm facing.

[image]

 - orange : Titan Aero X Carriage Front Plate MK3 V2
 - white : Titan Aero MK3 Part Cooling Fan Shroud MK3 R3
 - grey : X Carriage Universal Rear Bearing Plate MK3 V3

The 2 screws are M3x12, screwed into the heatsearts (in gold color)

Thank you for sharing this project. You have done an amazing work.

OmNomNomagon, 2020-03-06:
Great visualization. After this version I realized that this design was over constrained. The best bet is not to have heat-serts in the extruder plate (purple) but to only have them in the fan shroud. Then screw in as you have done here with 16-20mm M3 screws. This should pull everything together tightly.

pgrandin, 2020-03-06:
Makes sense!
Thank you!

closed 2020-03-06 by pgrandin