OmNomNomagon/ReDuplicator-MK3x #10: BOM clarity?

rogerschlachter, 2020-04-14:
Found my way here after doing some upgrades to my Wanhao i3+ (Hemera, Duet wifi, etc).  Pretty awesome project! Thank you so much for documenting everything you have and providing all of the STLs in a single pretty easy to find place.

I know it might be difficult to do, but would it be possible to clear up the BOMs a little?  For instance, I started going through what I'd need to build the Hemera version with the MK52 Heat Bed and things seem a little off?  For instance, the MK52 instructions say four bearings, but I see only three in the assembly of the bed, and I'd still need some for the X axis I imagine, vs say the extrusions where it tells me they replace the original parts. The MK52 instructions also leave off the Y idler part from the printed parts. And they mention hotend pieces so I'm not sure if it's trying to be all encompassing or just talk about the base parts?  It's just a little hard to get a clear picture of what I'll need. I don't want to disassemble my printer until I know I have everything I need printed.

Just wondering if there's maybe a better way to show the differences?  Maybe the table with all of the printed parts for the base version and a table with all of the purchased parts.  Then copy that same table into the altered builds and add a column for that option that says "Use Part X instead" or "Not used for this option".  Maybe something like this?

[image]

If the lists are clearer, I'd be up for coding up a little javascript to allow people to pick options and let it build a single list for them.

OmNomNomagon, 2020-04-24:
Thanks mate, sounds like a solid idea. I'll go through everything this weekend and clean it up

OmNomNomagon, 2020-05-18:
The 4 bearings mentioned in the instructions are for the x axis end. They are mentioned as they are additional parts required on top of the original wanhao parts. 2 bearings for into the X axis idler and 2 for the x axis motor mount. . The 3 heatbed carriage bearings and 2-3 x-carriage bears can be pulled accross from the original wanhao. Although you may as well upgrade them while doing this project. I'm going back through everything to ensure its all covered. I'll come up with a Full BOM just for the MK52 upgrade

closed 2020-08-04 by OmNomNomagon