OmNomNomagon/ReDuplicator-MK3x #12: Printed parts don’t fit with Misumi T-Nuts

Abai, 2020-08-06:
[image]
[image]

The printed parts that attach to extrusions are tabbed. So are the Misumi T-nuts. The slots in the printed parts tabs, need to be 14mm wide, 7mm on each side of the screw hole center.

I saw that you created a new revision of parts two days ago, I would like to fix these before reprinting. Could you provide the fusion 360 files? I can also make a pull request with fixed STL’s if you like.

Thanks!

Abai, 2020-08-06:
Alternative to making 14mm slots in printed tabs, it is possible to make all printed tabs 1mm high maximum. This is what misumi does with the tabs of gheir corner brackets.

Here are the drawings:

[image]
[image]
[image]

OmNomNomagon, 2020-10-18:
Hi mate, sorry you are having issues. M5 drop in T-nuts were always the preferred and recommended part for all the printed parts.  I recommend that you switch to drop in nuts for the printed parts and use the misumi ones for all the other structural parts.

closed 2020-10-18 by OmNomNomagon