OmNomNomagon/ReDuplicator-MK3x #14: New Mk52 X Belt Tensioner

cpolera, 2020-09-27:
I'd like to make a suggestion. I would flip around the holes for the pulley holder for the x axis. Currently when you put the inserts into it they are on the side being pulled towards. This means they can be pulled out if they arent gripping well to the plastic. I'd also be concerned about it maintaining tension over time. If you could put them in from the opposite side, then they'll always be pulled towards the "center" of the printed piece with no risk of pulling through or slowly slipping out over time.

OmNomNomagon, 2020-10-18:
HI mate, this was my original idea, however i went through a dozen or so prototypes before settling on this design. To insert them from the other side would require making the side and top walls a little thinner, or moving the m3 screw holes closer together. 

When the M3 holes were closer together I found it wasn't as stable. Making those walls thinner is also not an option as that is the current weak point in the design. I tested with overkill high torque steppers doing fast moves to stress the part. Every single time it would snap around the M5 bolt holding the idler well before pulling out the heat-sert.

I made a compromise to insert the heatserts from the back, in order to make the walls around the idler as thick as possible. Also I don't know if you've tried to remove heatserts from printed parts before, but it is very hard to remove them.

closed 2020-10-18 by OmNomNomagon

cpolera, 2021-01-16:
Sorry I didnt get back to this sooner. Thats a fair argument. I'd say that my issues may also be in part due to using PETG and the inserts that are not angled along the knurls. 

I'm reprinting in ASA now to see if that helps any.