OmNomNomagon/ReDuplicator-MK3x #15: Alternative supplies for heat inserts?

gawainXX, 2020-09-29:
I've scoured several vendors for the heat inserts in the BoM and haven't had any luck. All of the vendors I've come across didn't ship to the USA or the one that did had them on back order.

I've tried searching for other M3 inserts but none of the ones that I could find had a close outside diameter.

Could you recommend a supplier for the inserts you reference or a compatible alternative that delivers to the USA?

MarMed, 2020-09-29:
I purchased from here awhile back with no issues, and it looks like they
are in stock

https://www.newark.com/tr-fastenings/m3-uhbrhesf/brass-insert-unheaded-m3/dp/46Y7942

OmNomNomagon, 2020-10-18:
Thanks Marmed, [Octopart](https://octopart.com/m3-uhbrhesf-tr+fastenings-58189465) also has a good list of available suppliers

closed 2020-10-18 by OmNomNomagon

monroewilliams, 2021-03-29:
I found something similar on Amazon, which might be easier for some folks in the U.S. https://www.amazon.com/gp/product/B087NBYF65/