OmNomNomagon/ReDuplicator-MK3x #16: New designs improvements for standard hotend, frame

beriberikix, 2020-10-18:
Hello! I printed the original design (all standard core parts) around mid 2019 and have been tracking the changes like cooling fan clearance, x idler, cable connectors, etc. Do you plan to update the "standard" setup eventually?

Thanks!

OmNomNomagon, 2020-10-19:
Thanks for following the project mate. The answer is yes. I've gotten everything working perfectly on the current MK3s / hemera / mk52 version and I'm currently backporting everything to the original frame. I'm not sure on timelines sorry as I also have to convert from tinkercad to fusion for a lot of the parts and its a pain to use with the new file restrictions.

beriberikix, 2020-10-19:
Excellent, that's great to hear! I'm currently using an MK10 (Microswiss all-metal hotend) and would be happy to do any test prints if you need!

beriberikix, 2020-10-24:
What designs can be used with stock without having to do the tedious work of redoing in fusion 360? It seems like the x gantry idler and motor mount will just work? Perhaps putting together a checklist of what's done and what's not done yet would be useful?

beriberikix, 2020-11-01:
FWIW I tried printing the Hemera fan shroud, since it fixes the angle issue, but the mounting hole spacing is different :( I may resort to hacking the model up but I'm not well-versed in Tinkercad (F360 fan myself.)

donjakobo, 2021-02-01:
Any option to share the fusion 360 files? I've mounted a BMG-Wind clone (left handed I believe), replacing the titan aero that was giving me horrible grinding of filament (i wanted a dual geared solution).  Not finding any compatible fans for the new hot end placement (as it shifts slightly up and to the right vs titan aero).

If I could get the Fusion 360 files I would love to take a stab at modifying the existing fan to adjust it.

MarMed, 2021-02-04:
BMG-Wind would be a very welcome addition imo

donjakobo, 2021-02-08:
I've gone and done my best messing with the TinkerCad objects reference to make a workable prototype for the **BMG-Wind** (left handed I believe is the name) that directly replaces the Titan Aero.  I'm going through a coup iterations to get it just right, and just elongate and extend the Titan Aero cooler (rear mounted). 

**Key issues:**

1. BMG-Wind Hotend tip is higher (at least for me) than the titan aero, so need to 'shrink' the height of cooler down so it isn't impacting bed
2. BMG-Wind hotend tip is offset so moving the cooler further right and elongating to the front.

OmNomNomagon, 2021-02-11:
> What designs can be used with stock without having to do the tedious work of redoing in fusion 360? It seems like the x gantry idler and motor mount will just work? Perhaps putting together a checklist of what's done and what's not done yet would be useful?

The X gantry parts wont work unfortunately as the rod spacing is different between both printers.  I'm hoping to have a version which will work on both.

xiaoyue, 2021-03-28:
I'm thinking about building the new MK52 frame design, but to keep using the old/original bed and y-carriage. Would it be possible by simply adjusting the Y rod spacing in the new design to be compatible?

OmNomNomagon, 2021-04-07:
> I'm thinking about building the new MK52 frame design, but to keep using the old/original bed and y-carriage. Would it be possible by simply adjusting the Y rod spacing in the new design to be compatible?

Yes, you can use the original heatbed with the MK52 frame design. however you can't use the MK52 bed with the original frame.

iZz0s, 2021-05-25:
are the extrusions for the MK52 edition the same as listed in the BOM?

i've built already a normal core edition and was thinking to reprint the MK52 parts but dont know if it fits with the BOM extrusions, i haven't found other infos.