OmNomNomagon/ReDuplicator-MK3x #18: Cable Guide Alternate - Motor Mount 360 doesn't fit my X motor

monroewilliams, 2021-03-29:
This part doesn't fit my motor at all. The outside of the motor is approximately 42mm square, and the inside of the square part is about 35mm. 

This is definitely the original motor that came on my printer (Monoprice Maker Select v2, purchased in November 2016), but as you can see in the photos it didn't come with a heat sink like the ones shown in the assembly guide. It looks like my motor is a NEMA 17, but this part was designed for a NEMA 14?

This is especially odd since it looks like the hole spacing on the X motor mount I've printed (`X Gantry Motor Mount - MK52 Expansion 360 v2`) seems to have the correct hole spacing for a NEMA 17 motor (31mm).

[image]
[image]

monroewilliams, 2021-04-01:
I see that someone already did a remix to address this:

https://www.thingiverse.com/thing:4590435

so I'll consider this issue resolved.

closed 2021-04-01 by monroewilliams

OmNomNomagon, 2021-04-07:
Thanks for bringing this up mate. I triple checked the part and it looks like I missed a couple of cut operations before i exported the part. I'll have the fixed one uploaded shortly